greatscottgadgets/hackrf #1546: HackRF One blinking 3 diodes Tx, Rx and USB, hardware issue

otpisani, 2025-04-22:
### What outcome were you hoping for?

What is usually hardware failure regarding blinking 3 diodes Tx,Rx and USB, main NXP Arm CPU, or something else, issue is that I change that IC LPC4320FBD144, but diodes are still blinking. I'm not pro regarding soldering, but I check everything twice. 
Difference between working HackRF and this one is resistance in continuity mode at testing point P79 on HackRF one in failure mode, and working card has 1. If someone knows what else can be suspicious, please let me know

### What outcome actually happened?

When you plug in USB cable into PC, hackrf one start blinking 3 diodes

### What operating systems are you seeing the problem on?

Linux

### What is the output of ```hackrf_info```?

Card can be seen into DFU mode, but after download flash, device is gone.

### Are you using any third-party software?

Yes

### Are you using any third-party hardware?

No

miek, 2025-04-23:
There are two ways that error state can come up:
* The firmware doesn't match the detected platform. If you flashed either the `jawbreaker` or `rad1o` firmware files to a HackRF One it would happen. Also it'd happen if an R9 board was flashed with older firmware that didn't support R9. It may also happen if the pins involved in platform detection aren't soldered properly, so it's detecting the wrong platform.
* The firmware is unable to load the CPLD SRAM at boot. That might be caused by bad connections on the pins between the CPU and CPLD, or if there is other hardware damage related to whatever damaged the MCU in the first place.

First just double check you're flashing the right firmware file. If that doesn't work, what I'd recommend is narrowing down the cause: You could attach a debugger and see how it's getting into the `halt_and_flash` function. If you're not set up for that, it might be easier is to just modify the firmware and increase the duration on one of the calls by 10x, program it, and see if the flashing got slower.

Here's the platform detection code:
https://github.com/greatscottgadgets/hackrf/blob/e5cfe1ac2b4d0621705ba3006d0f8abd447f50a3/firmware/common/platform_detect.c#L186-L214

and the CPLD loading code:
https://github.com/greatscottgadgets/hackrf/blob/e5cfe1ac2b4d0621705ba3006d0f8abd447f50a3/firmware/hackrf_usb/hackrf_usb.c#L251-L253

otpisani, 2025-04-23:
I have hackrf r9, so if I get you correct I need to change 

` case HACKRF1_R9_RESISTORS: 
 	if (!(supported_platform() & PLATFORM_HACKRF1_R9)) { 
 		halt_and_flash(3000000); `

into

` case HACKRF1_R9_RESISTORS: 
 	if (!(supported_platform() & PLATFORM_HACKRF1_R9)) { 
 		halt_and_flash(30000000); `

But what is correct parameter when you compiled firmware, and you want dfu file, I can see in dfu-util.cmake this line. 

`else(NOT DFU_NOT_FOUND)
        MESSAGE(STATUS "dfu-suffix not found: not building DFU file")
endif(NOT DFU_NOT_FOUND)
`
cmake -DBOARD=RHACKRF1_R9 xyz

miek, 2025-04-23:
If you're getting that error when building, then you need to install `dfu-util` which includes the `dfu-suffix` program.

miek, 2025-07-03:
I'm going to close this as there hasn't been a response in a while, but feel free to re-open or open a new issue if you still have more questions about this.