greatscottgadgets/hackrf #1547: Extreme 977Hz 480MhZ RFI?

Degamisu, 2025-04-23:
### What outcome were you hoping for?

Clean (or almost clean) UHF band

### What outcome actually happened?

A huge spike in the UHF band, causing harmonic interference to most of the radio spectrum. This was spotted 4 days ago. I tried different locations, still occurs. My RTLSDR (no issues) can pick it up, not as much as HackRF, which leads me to believe its internal.

After listening to it using AM, the signal outputs a `977hz` tone which is a clock rate of some analog oscillator

## NOISE

[image]

## SWEEP (100MhZ - 600MhZ)

>[!NOTE]
> Notice the spikes in the signal as a result of the harmonics/

[image]

### What operating systems are you seeing the problem on?

Windows 11, WSL, Debian, Any OS.

### What is the output of ```hackrf_info```?

```bash
hackrf_info version: 2023.01.1
libhackrf version: 2023.01.1 (0.8)
Found HackRF
Index: 0
Serial number: 000000000000000066a062dc274c519f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x004b475b
Hardware Revision: unknown
Hardware appears to have been manufactured by Great Scott Gadgets.
Hardware supported by installed firmware:
    HackRF One
```

### Are you using any third-party software?

`SDRPP`, `HackRF-Sweep-Spectrum`

### Are you using any third-party hardware?

`27MhZ Telescopic Antenna (Monopole)` + `Stock Antenna`

miek, 2025-04-23:
At that frequency, it's most likely to be coming from the USB lines, but it shouldn't be that high. It will depend heavily on the USB cable used and it's quite common to find ones that don't have any shielding, so I'd double check that it is shielded and/or try some other cables to see if you see any difference.

Degamisu, 2025-04-27:
@miek Thought it was an issue with USB. I was using the stock USB for the stock GSG, I'll try it with some other mini USB cables. Perhaps an Xbox controller charging cable will work? I'll post the results.

Also: The signal does change when I wiggle it around a little. Definetly the USB.

Degamisu, 2025-04-27:
As I suspected, a charging cable _obviously_ cannot send over data 😆

Degamisu, 2025-06-20:
Of course! My gain settings are:

LNA: `40dB`
VGA: `24dB`

No amp, no BT.

Frequency is `480 MHz` 
Sample rate = `8 msps + 1.75M Bandwidth Filter, Decimated 4x.`