greatscottgadgets/hackrf #1549: hackrf_sweep sometimes doesn't output result  on Raspberry Pi Zero 2W

targetod, 2025-05-01:
### What type of issue is this?

intermittent - occurring irregularly

### What issue are you facing?

We reproduced bug 'hackrf_sweep sometimes outputs results, and sometimes it doesn't' on Raspberry Pi Zero 2W. 
Pretty similar to https://github.com/greatscottgadgets/hackrf/issues/1052  (it was fixed and closed).

Probably it's the same as  https://github.com/greatscottgadgets/hackrf/issues/1414 

### What are the steps to reproduce this?

I use sweep_callback in my own service, 
but it should be reproduced  using 
`while true; do hackrf_sweep -1 -f 2400:2500; done > out 2>&1`

### Can you provide any logs? (output, errors, etc.)

With some logging, we can see that there is an offset ( transfer was off by one packet somewhere).
```
HackrfDevice::sweep_callback(): Not Frequency 0x7F (127); buf[0]: 255 buf[1]: 251
[01/17/25 12:43:57 +02:00] [info] 
HackrfDevice::sweep_callback(): FF FB FF FB FF FB FF FA FE FB 00 FB FF FB FF FB 
7F 7F 00 D2 49 6B 00 00 00 00 FF FB FF FB FF FB
```
Operating system:

```
pi@rasp:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
hackrf_info output:

```
hackrf_info version: 2024.02.1
libhackrf version: 2024.02.1 (0.9)
Found HackRF
Index: 0
Serial number: 0000000000000000a31c64dc2160920f
Board ID Number: 2 (HackRF One)
Firmware Version: 2024.02.1 (API:1.08)
Part ID Number: 0xa000cb3c 0x004f475a
Hardware Revision: older than r6
Hardware supported by installed firmware:
    HackRF One
```
libusb version:
`ii  libusb-1.0-0:arm64                   2:1.0.26-1                       arm64        userspace USB programming library `

targetod, 2025-05-02:
Hi @martinling,  please pay attention.
Would you happen to know how quickly it can be fixed?

martinling, 2025-05-02:
If I knew what was causing the problem, I'd be able to estimate how quickly it could be fixed. But I have no idea what causes this.