greatscottgadgets/hackrf #1552: Resetting the Cortex ARM?

kpelckma, 2025-05-21:
### What outcome were you hoping for?

Hi - we're learning how to program the Cortex M4/M0 on the HackRF, but ran into a catch22:  after happily getting a basic program (similar to the blinky example [here](https://github.com/greatscottgadgets/hackrf/tree/master/firmware/blinky)) to run on the ARM, we also lost access to the board:

[image]

How to reset the board to the standard firmware?

### What outcome actually happened?

(see screenshot)

### What operating systems are you seeing the problem on?

Linux/Ubuntu

### What is the output of ```hackrf_info```?

No HackRF boards found.

### Are you using any third-party software?

No.

### Are you using any third-party hardware?

No.

mossmann, 2025-05-21:
Do you have a PortaPack or any other expansion hardware connected to the HackRF?

kpelckma, 2025-05-22:
> Do you have a PortaPack or any other expansion hardware connected to the HackRF?

Nope.

[image]

miek, 2025-05-22:
One thing that could cause that error is a permissions issue. Could you try running `dfu-util` as root or with sudo?

kpelckma, 2025-05-22:
Nope. running under SUDO gives the same error ... it complains about an invalid DFU suffix signature?

miek, 2025-05-22:
Aha, I think I see the problem - you need to download `hackrf_usb.dfu` not `hackrf_usb_dfu.bin`.

kpelckma, 2025-05-22:
Aha - so this is not in my local build ... whats different and where can I get this on the WWW?

miek, 2025-05-22:
During the cmake configuration, you probably get a message saying "dfu-suffix not found: not building DFU file". You just need to install the `dfu-util` package, then re-run cmake, and it should generate it.

The difference is that the `.dfu` file has some extra header/footer with information parsed by the bootloader, but the `.bin` file is just the bare firmware.

kpelckma, 2025-05-22:
Yay - Thanks a bunch!

closed 2025-06-05 by miek