greatscottgadgets/hackrf #1559: LSUSB shows NXP, DFU appears to work, LSUSB shows nothing, then hackrf_spiflash fails

dfriesen007, 2025-06-06:
### What outcome were you hoping for?

I expect:
a) LSUSB to show the hackrf after a DFU download.
b) hackrf-spiflash to succeed.

### What outcome actually happened?

The HackRF One only shows up in LSUSB (as NXP) if I reboot with DFU button held.
After the DFU download, LSUSB doesn't show the device anymore.

<pre><span style="color:#26A269"><b>david@P5QL-debian</b></span>:<span style="color:#12488B"><b>~/Downloads/hackrf-2024.02.1/firmware-bin</b></span>$ lsusb
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 049f:0051 Compaq Computer Corp. KU-0133 Easy Access Interner Keyboard
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 025: ID 1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 046d:c52e Logitech, Inc. MK260 Wireless Combo Receiver
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
<span style="color:#26A269"><b>david@P5QL-debian</b></span>:<span style="color:#12488B"><b>~/Downloads/hackrf-2024.02.1/firmware-bin</b></span>$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 1fc9:000c
Device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        42312 bytes
Download done.
dfu-util: unable to read DFU status after completion (LIBUSB_ERROR_NO_DEVICE)
<span style="color:#26A269"><b>david@P5QL-debian</b></span>:<span style="color:#12488B"><b>~/Downloads/hackrf-2024.02.1/firmware-bin</b></span>$ hackrf_spiflash -w hackrf_one_usb.bin
File size 42248 bytes.
hackrf_open() failed: HackRF not found (-5)
<span style="color:#26A269"><b>david@P5QL-debian</b></span>:<span style="color:#12488B"><b>~/Downloads/hackrf-2024.02.1/firmware-bin</b></span>$ lsusb
Bus 008 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 002: ID 049f:0051 Compaq Computer Corp. KU-0133 Easy Access Interner Keyboard
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 046d:c52e Logitech, Inc. MK260 Wireless Combo Receiver
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
</pre>

I should add, after the DFU download, I have power led solid green,
red tx and amber rx flashing 5 or so times a second.

### What operating systems are you seeing the problem on?

Linux Mint and Debian

### What is the output of ```hackrf_info```?

david@P5QL-debian:~/Downloads/hackrf-2024.02.1/firmware-bin$ hackrf_info
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)
No HackRF boards found.

### Are you using any third-party software?

Not yet. Hope is to use GNU Radio Companion on Linux Mint.

### Are you using any third-party hardware?

No.

dfriesen007, 2025-06-06:
Perhaps a semantic issue, but, maybe I've (you've) got upload and download backwards? I'm typing what I see online in the troubleshooting, but, it seems unconventional.

As I understand it:

download means request and receive a file from a remote device. ie: from server or peripheral (hackrf) to the PC (linux desktop or laptop) I'm typing on

upload means send a file to a remote device (ie: from PC to peripheral or server)

dfriesen007, 2025-06-07:
After running dfu-util, "download" 100%, 42312 bytes:

The 3v3 is on solid
The 1v8 is off.
The RF is off, but anode reads 1.4V DC
The USB is off, but anode reads 1.8V DC (it's maybe 5% bright?- a very faint solid glow)
The tx and rx are flashing with a period of approx 720 ms. (1.3888 Hz)

Measured with a 10x probe on an old analog HP 1740A scope.

mossmann, 2025-06-14:
Is this an older HackRF One? Has it been physically damaged or modified in any way? Can you please attach a photo of the board?

Are there any add-ons connected? If so, please remove them and try again.

dfriesen007, 2025-06-20:
Photo attached,  I hope. 
There's a couple coils? Pointed to by a pen, that look like they've seen better days.

[image]

dfriesen007, 2025-06-29:
-awating response

mossmann, 2025-07-17:
We have seen that particular inductor damaged by the batteries of poorly designed PortaPacks. If you have a PortaPack with a battery, you may need to do something to secure the battery in a position that does not contact the inductor. This would help prevent future damage if you repair or replace the HackRF.

closed 2025-07-17 by mossmann