greatscottgadgets/hackrf #1560: Lighting Strike

zglozman, 2025-06-09:
### What outcome were you hoping for?

I had a serrious lighting strike nearby, and it totally bricked my hackrf, no leds turn on when powered, can someone tell me where to start trouble shooting ?

### What outcome actually happened?

help

### What operating systems are you seeing the problem on?

Linux

### What is the output of ```hackrf_info```?

nothing

### Are you using any third-party software?

sdr

### Are you using any third-party hardware?

a

Moji14, 2025-06-10:
I hope this helps...
https://eshop.sectron.eu/en/lightning-arrestor-sma-m-sma-f/p-1371/

martinling, 2025-06-19:
If you're comfortable reading a schematic and probing with a meter then you could try to identify what's damaged.

If no LEDs are on then you don't have a 3.3V supply. That supply comes from the U21 regulator.

The 5V supply from USB flows through FB1 and Q5 to reach U21. It's also possible for U15 to fail short. Check the voltage at each component to see if the 5V supply is getting to U21.

If it is, then you may need to replace U21 and possibly other components associated with it.

Note however that a lightning strike is a serious event and there could be multiple damaged components, so even after recovering the power supplies the board may not be recoverable.

closed 2025-07-17 by martinling