greatscottgadgets/hackrf #1562: Hammond Box 1455J1201 Endplate Template

jlpoolen, 2025-06-21:
### What outcome were you hoping for?

The documentation at https://hackrf.readthedocs.io/en/latest/enclosure_options.html#enclosure-options provides as of 6/20/25:
Hammond 1455J1201: HackRF One fits this extruded aluminum enclosure and other similar models from Hammond Manufacturing. In order to use the enclosure’s end plates, you will have to drill them. An end plate template can be found in the HackRF One KiCad layout.
Try as I did using KiCad 9.0, i could not find the template.
I'd like to order the box from Digikey, but I don't want to mess it up with guestimate drilling.  

My unit was purchased from Adafruit in June, 2025.  There are no marking on the case to identify which version, and I recall reading that the PCB has the version, but from the documentation "Opening the plastic case of your HackRF One will most likely destroy the tabs that hold it together..." and I was not able to find any instruction on how to open the case without damaging it.

Can a PDF of the template be staged and/or can you tell me which KiCad file within the project I should be referencing?

### What outcome actually happened?

Cannot find template for drill holes.

### What operating systems are you seeing the problem on?

NA

### What is the output of ```hackrf_info```?

NA

### Are you using any third-party software?

NA

### Are you using any third-party hardware?

NA

martinling, 2025-06-21:
The templates are in the `User.Comments` layer of `hardware/hackrf-one/hackrf-one.kicad_pcb`, you may need to turn that layer on to see them.

[image]

jlpoolen, 2025-06-21:
Thank you, that was very helpful and I found the layer. 

[image]

I am trying to give something back by providing the dimensions one would need to properly drill, but I'm finding my inexperience in KiCad is causing doubt about the accuracy of what I am trying to do.  I assumed everything is in metric, and I found if I changed the snap to 0.25 mm I appeared to get on center of points and lines, but then I started to get off, so I changed snap to 0.2 mm.  But then interdistances between the LEDS did not align, so I come back here asking if the drawing should have whole or reasonable fractions, e.g. 1/2 or 1/4, millimeter precision, or was a design from Hammond overlayed and then the circles from the PCG brought up into the User.Comments layer.  If I'm not right on the mark, then my contribution could be worse than none at all if the holes do not line up just right.

[image]